brett8883/DJI_Super-Patcher #86: Is it safe to flash "Upgrade_Custom_FC_MavicPro_to_01.04.0500_NO_305_NO_306_NO_905_NO_801_dji_system.bin"?

DavidZang, 2020-02-15:
Hi, thank you very much for providing such a grest tool for us. I have a question on my Mavic Pro Platinum with firmware V01.04.0300. The latest firmware is V01.04.0500 and the DJI GO app always reminds me to update. There is a solution which I see from other issuse that flash this file "Upgrade_Custom_FC_MavicPro_to_01.04.0500_NO_305_NO_306_NO_905_NO_801_dji_system.bin". I'd like to know if it will lead to other unknown problems?
Thank you very much.

brett8883, 2020-02-16:
Yes you can upgrade to 01.04.0500 with that firmware. I made that file as well and it’s what I use on my Mavic Pro Platinum. One of these days when I have more time I will add it to Super-Patcher. 

Update the remote to 01.04.0500 as well so the whole system will be on the latest firmware which is necessary to stop the update prompts. 

You must do Super-Patcher on 01.04.0300 and THEN upgrade to 01.04.0500 with the file you reference above.

DavidZang, 2020-02-17:
Thank you very much. I upgrade the firmware to 01.04.0500 with SP mods kept. :)

closed 2020-02-17 by DavidZang

sober607, 2020-10-04:
But this file is real core of 0500 firmware? Or just 0300 fiurmware with patched headers of firmware version but old core from 0300?

shahdevi81, 2020-10-04:
I want to upgrade nfz and altitude height....can u show me step by step to
me...i want to do it with my computer...