brett8883/DJI_Super-Patcher #87: mavic pro platinum Hover Drifting Problem

kostask1, 2020-02-20:
thanks a lot for the patch but i have a little problem here..
in version 01.04.0500. the drone fly normally without drifting or instability.
after run the patch (with no problem.everything good) drone has instability and drift without
move the sticks. both in height and length.
1.when it is in hover the drone drift.
2.it writes me on the screen 56m high and in reality it is 4 meters.20-22 satellates less.
when I put it on again 01.04.0500 with dji assistant work normaly and without problems.no drift,no instability.
run second time the patch without issus (downgraded 0300 first with patch) and i have again the same problem...
i calibrate everything..any solution??

norfeas, 2020-02-20:
Hi. I have exactly the same problem with mavic pro and 1.04.0300.  I made all the process from the beginning also I did IMU and compass calibration and it seems to have improved. I have to make some more tests. When i was with mixing modules everything was fine.

brett8883, 2020-02-22:
You can upgrade to 01.04.0500 using this process  

https://github.com/brett8883/DJI_Super-Patcher/issues/84#issuecomment-585486716

kostask1, 2020-02-22:
i upgrade to.0500 but no luck...
calibrate again everything.no luck.
Super-Patcher mods works normally.
when i run again 01.04.0500 with dji assistant works perfect.
confused..

brett8883, 2020-02-23:
What are the conditions when you are testing? Had someone mention this issue before and it turned out he was testing at night.

kostask1, 2020-02-23:
yes i testing at night.problem with that?no wind at all.
the first time with 0300 i testing daylight.
no issue before 
I will continue tests today

kostask1, 2020-02-23:
i did the process from the beginning
1.run the patch with 0300(I didn't activate boost frequency mode this time)
2.i upgrade to.0500 with the patch
3.calibrate again everything
4.flight test, no wind,daylight, works normally with a little drift (no 1-2 meters discrepancy as in the beginning)
5.I'll try again and report it.
thanks a lot brett8883

norfeas, 2020-02-23:
I also did the process from the beginning. Upgrade to 01.04.0500, calibrate again everything. Flight test was good as @kostask1 said  a difference of 1-2 meters in height.

kostask1, 2020-02-23:
activate boost frequency mode ?
where are you from @norfeas ?if I'm not curious

norfeas, 2020-02-23:
Yes boost mode works fine and i fly with modeed dji go 4  4.1.22. I am from Greece.

kostask1, 2020-02-23:
send me if you want email to kostaskex@hotmail.com

brett8883, 2020-02-23:
> yes i testing at night.problem with that?no wind at all.
> the first time with 0300 i testing daylight.
> no issue before
> I will continue tests today

The drone uses VPS for precision hovering. GPS isn’t accurate enough to hold precise hover. At night VPS doesn’t work so the drone will drift

brett8883, 2020-02-23:
This is why there was no drifting when you tested the .300 during the day and had drifting when you tested Super-Patcher at night.

kostask1, 2020-02-24:
I continued the tests.
I'm sure the problem is satellites(galileo gps).
flight test, no wind,daylight, works normally with NO drift  with 10-12 satellites.works perfect.
when they go up (20-22 satellites) the drone drifting and i have instability.i am sure for this 99,9%
tests 3 batteries today.every time the same result.
when activated over 20 satellites it has no stability.
in vision mode (atti) the drone is rock.stability its perfect.
it is possible for you to mode the patch without activate the galileo gps?
and i send you a gift for you @brett8883

norfeas, 2020-02-25:
Upon continued tests, also believe that the problem comes from the addition of extra satellites, as @kostask1  mentioned. We have exactly the same problem and the same symptoms. Mavic pro works normally with NO drift with 10-12 satellites, when the number of satellites is 18-22 the drone drifting up and down 2 or 3 meters and make eratic moves as if trying to correct his position without any reason to do so.
Is it possible to do a firmware without galileo so we can flash it and test it.
Thank you in advance!

yingxiang123, 2020-02-28:
> You can upgrade to 01.04.0500 using this process
> 
> [#84 (comment)](https://github.com/brett8883/DJI_Super-Patcher/issues/84#issuecomment-585486716)

As you say, Mavic Pro has to be running stock V01.04.0300 fw for all modules , I'm following your method， super_patch goes well  01.04.0300 ，does  super_patch work 01.04.500? how to  upgrade to 01.04.0500 ? should  i  mix firmware modules and flash them ?

brett8883, 2020-03-04:
> As you say, Mavic Pro has to be running stock V01.04.0300 fw for all modules , I'm following your method， super_patch goes well 01.04.0300 ，does super_patch work 01.04.500? how to upgrade to [...]

The comment you referenced has the directions for upgrading to 01.04.0500 after you have done SP on 01.04.0300. The method it uses is the same concept as mixed modules yes, however, the flight controller module in 01.04.0500 is the EXACT same module as in 01.04.0300 same MD5 and all. So even though we are using the exploit that was originally used for mixing old flight controller modules with other modules from newer firmware, namely the fact the Mavic Pro will not upgrade or downgrade modules that don’t exist in the firmware being flashed, once you have flashed the modified 01.04.0500 that I made you end up with the same modules that are in the stock 01.04.0500 but still get the benefit of the super-patcher modified flight controller. 

You could actually flash stock 01.04.0300, upgrade to 01.04.0500 using the modified 01.04.0500 from the comment you reference above giving you a stock 01.04.0500 without the lockout mechanism than would other wise prevent Super-Patcher from working and THEN do Super-Patcher. 

The reason I say to do super-Patcher on 01.04.0300 first and then upgrade is because then the DJI Go 4 app will correctly show you as being on 01.04.0500 and won’t ask up to upgrade anymore because it can see that you are already on the latest firmware.