brett8883/DJI_Super-Patcher #89: Remove height limitations

maulanarizqi, 2020-02-23:
halo there,

i set height limitation on 500 meter but when it reach 120 meter there was red notification, mentioned that limitation only 120, but i managed to 180m with that notification, but it quite annoying, while in 1.400 firmware 250m height without red notification....

maulanarizqi, 2020-02-23:
anyone have Assistant 2 1.1.2 link for download ?

knarfboy, 2020-02-23:
Google has the link....

maulanarizqi, 2020-02-23:
> Google has the link....

 google has valid link ? did you check in real google site ?

knarfboy, 2020-02-23:
421000 Hits
https://www.google.de/search?sxsrf=ALeKk00LxvD48enusYiMGzmgM467WVYxJg%3A1582473139179&source=hp&ei=s59SXuXzB4WZkgWC_ZuoDQ&q=dji+assistant+2+1.1+2+download&oq=Assistant+2+1.1.2&gs_l=psy-ab.1.1.0i19j0i22i30i19l3j0i333l2.1430.1430..6185...1.0..0.96.96.1......0....2j1..gws-wiz.YcLDXJraruA

maulanarizqi, 2020-02-23:
> 421000 Hits
> https://www.google.de/search?sxsrf=ALeKk00LxvD48enusYiMGzmgM467WVYxJg%3A1582473139179&source=hp&ei=s59SXuXzB4WZkgWC_ZuoDQ&q=dji+assistant+2+1.1+2+download&oq=Assistant+2+1.1.2&gs_l=psy-ab.1.1.0i19j0i22i30i19l3j0i333l2.1430.1430..6185...1.0..0.96.96.1......0....2j1..gws-wiz.YcLDXJraruA

where is google download link for Assistant 2 1.1.2 ?

knarfboy, 2020-02-23:
omg
->https://dji.retroroms.info/howto/download_assistant

brett8883, 2020-02-23:
> halo there,
> 
> i set height limitation on 500 meter but when it reach 120 meter there was red notification, mentioned that limitation only 120, but i managed to 180m with that notification, but it quite annoying, while in 1.400 firmware 250m height without red notification....

This wasn’t using Super-Patcher I assume.

maulanarizqi, 2020-02-23:
> This wasn’t using Super-Patcher I assume.

im using your superpatcher, i notice that only fcc boost and galileo satellite gained until 27, but the flight height and nfz still no affect... what do you suggest ?  re patcher the drone ?

brett8883, 2020-03-04:
> im using your superpatcher, i notice that only fcc boost and galileo satellite gained until 27, but the flight height and nfz still no affect... what do you suggest ? re patcher the drone ?

What do you mean? Have you tried flying above 500 meters?

maulanarizqi, 2020-03-13:
not yet, with red notification appeared on DJI GO app screen, im affraid to fly higher, and then the nfz still appeared, some area not allowed to fly, i already repatcher, but still no affect, im trying to downgrade into superpatcher 2.0

maulanarizqi, 2020-03-13:
> What do you mean? Have you tried flying above 500 meters?

not yet, with red notification appeared on DJI GO app screen, im affraid to fly higher, and then the nfz still appeared, some area not allowed to fly, i already repatcher it, but still no affect, im trying to downgrade into superpatcher 2.0

Weirdoz, 2020-03-13:
It doesn‘t matter what the app will show you, the AC will fly above 500m and inside an NFZ, because the AC will ignore all this nag screens and notifications. If you set your hight limit to 30m and the AC fly above it, then you can be sure it will also ignore the 500m limit. The 120m nagscreen and the NFZ also belongst to local limitations and comes from the app, not the drone.
Fly save

brett8883, 2020-03-19:
> not yet, with red notification appeared on DJI GO app screen, im affraid to fly higher, and then the nfz still appeared, some area not allowed to fly, i already repatcher it, but still no affect, im [...]

I was quite explicit in the read me that the NFZ warnings in the app will still appear. DJI has also added warnings when flying over 400 feet in the latest update. That’s coming from your app not your drone. The drone will fly wherever you want it to.

closed 2020-03-20 by maulanarizqi