brett8883/DJI_Super-Patcher #92: Mavic pro firmware version still show the 01.04.0300 but not N/A or 0000000

ywslodl, 2020-02-27:
I just patched my Mavic pro and I found that almost the last step, the firmware version I checked still showed as 01.04.0300 but not N/A or 0000000. For my friends' phantom 4 it worked showed N/A in there. Therefore I have tried the height still 499m which failure on my Mavic pro. Is there which step I made it wrong? Than you

13316179943, 2020-02-27:
I think you hadn't flash the firmware into the drone successfully. If it is, your drone will restart.

ywslodl, 2020-02-27:
Thank you for your reply. Yes the drone didn't start itself as the Flashing procedure starts from 15% and finished at 100% only 2 seconds. Therefore I manually started the drone however didn't work. What if I can try your v1.7 offline patcher as the network here probably blocked by the firewall. As I spent lots of minutes on waiting the first '2.0.zip' only.

13316179943, 2020-02-28:
> Thank you for your reply. Yes the drone didn't start itself as the Flashing procedure starts from 15% and finished at 100% only 2 seconds. Therefore I manually started the drone however didn't work. [...]

Well, I met the same problem. I think the problem is on the server.  The download problem made the program can't run well, because it skip the download firmware step. Luckily, I download all the things it needs at last after several tries. After all the things download, I revised it code to made it run without network. By this way, I can use it without wait a lot of time. If your have downloaded the '2.0 zip', you can have a try.

ywslodl, 2020-02-28:
> Well, I met the same problem. I think the problem is on the server. The download problem made the program can't run well, because it skip the download firmware step. Luckily, I download all the [...]

Did you use the V1.7 offline version and just v2.0 version? Because I just tried probably the firewall here and I stuck on waiting the “2.0.zip”. As my friend told me he opened that file in HK only within 20 seconds and then in the page of choosing the aircraft model. I will try to let him copy the whole folder and send to me afterwards.

May I also ask do you still need to amend the parameter such as height limit, NFZ after installation? Because I saw some YouTuber they will change the parameter such as 10000m in height limit but the read document here said that no additional action required to enable those features. And I checked my DJI assistant debug version and I found that the height isn’t 10000 metre of course. Do I need to amend that to 10000m to enable the height limit? Or just I can fly higher than 500m? Thank you

13316179943, 2020-02-28:
> Did you use the V1.7 offline version and just v2.0 version? Because I just tried probably the firewall here and I stuck on waiting the “2.0.zip”. As my friend told me he opened that file in HK only [...]

I just use the V2.0 version. It's sure that the program can achieve all the function it describe without amend the parameter. But it can't help you to change the speed. What's more, the 2.0 version will start the download process when you restart it. No matter you have download the '2.0 zip' or not. So you still need to do on the code to make it work without network. The other solution is open the same program after the one is starting download '2.0 zip'process, then the other program will skip the download '2.0 zip' process. The presupposition is you have prepare all the things it needs.

ywslodl, 2020-02-28:
I just patched 

> I just use the V2.0 version. It's sure that the program can achieve all the function it describe without amend the parameter. But it can't help you to change the speed. What's more, the 2.0 version [...]

I just patched my Pro on my friends' Win 10 using the v2.0 and the final step showed that the firmware is 00.00.0000. Actually I though it works however the height limit still 499m I tried on the simulator within Dji Assistant 2 v.1.2.5 (official one not debug one) which didn't pass the height limitation. Because here I am not able to fly shortly probably I will try later. 

I also tried that v1.7 offline which I didn't pass the final step of showing the NLD, and my FC showed 03.02.44.07 but not the target of showing 03.02.44.10.

KeMing541, 2020-02-28:
> I just patched
>
>
> I just patched my Pro on my friends' Win 10 using the v2.0 and the final step showed that the firmware is 00.00.0000. Actually I though it works however the height limit still 499m [...]
Congratulations! You have already flash the firmware into the aircraft. Your aircraft can flight higher than that in the simulation.

ywslodl, 2020-02-28:
Excuse me?

ywslodl, 2020-02-28:
> I just patched
>
>
> I just patched my Pro on my friends' Win 10 using the v2.0 and the final step showed that the firmware is 00.00.0000. Actually I though it works however the height limit still 499m [...]

Excuse me?

KeMing541, 2020-02-28:
Ah，I just a senior high school student from China. I don't know if we have communications gap. If you have any question, please don't hesitate to let me know. At the same time, this can help me to develop my English skills, which take an important place in the college entrance exam, I think you should know.

ywslodl, 2020-02-28:
> Ah，I just a senior high school student from China. I don't know if we have communications gap. If you have any question, please don't hesitate to let me know. At the same time, this can help me to [...]

I am sorry. Is there anything I can help you or you can improve the communication relating to the patch here?

KeMing541, 2020-02-28:
> I am sorry. Is there anything I can help you or you can improve the communication relating to the patch here?

Sure！It a good things to communicating with others while I can only learning on the internet in home because of the virus!

KeMing541, 2020-02-28:
Sorry for you don't know my meanings. If you can't understand it, you can ignore the reply.

brett8883, 2020-03-06:
If you are having trouble please share your log file so we can figure out what your issue is.

brett8883, 2020-03-09:
> > I just patched
> > > > > > Thank you for your reply. Yes the drone didn't start itself as the Flashing procedure starts from 15% and finished at 100% only 2 seconds. Therefore I manually started the drone however didn't work. What if I can try your v1.7 offline patcher as the network here probably blocked by the firewall. As I spent lots of minutes on waiting the first '2.0.zip' only.
> > > > > 
> > > > > 
> > > > > Well, I met the same problem. I think the problem is on the server. The download problem made the program can't run well, because it skip the download firmware step. Luckily, I download all the things it needs at last after several tries. After all the things download, I revised it code to made it run without network. By this way, I can use it without wait a lot of time. If your have downloaded the '2.0 zip', you can have a try.
> > > > 
> > > > 
> > > > Did you use the V1.7 offline version and just v2.0 version? Because I just tried probably the firewall here and I stuck on waiting the “2.0.zip”. As my friend told me he opened that file in HK only within 20 seconds and then in the page of choosing the aircraft model. I will try to let him copy the whole folder and send to me afterwards.
> > > > May I also ask do you still need to amend the parameter such as height limit, NFZ after installation? Because I saw some YouTuber they will change the parameter such as 10000m in height limit but the read document here said that no additional action required to enable those features. And I checked my DJI assistant debug version and I found that the height isn’t 10000 metre of course. Do I need to amend that to 10000m to enable the height limit? Or just I can fly higher than 500m? Thank you
> > > 
> > > 
> > > I just use the V2.0 version. It's sure that the program can achieve all the function it describe without amend the parameter. But it can't help you to change the speed. What's more, the 2.0 version will start the download process when you restart it. No matter you have download the '2.0 zip' or not. So you still need to do on the code to make it work without network. The other solution is open the same program after the one is starting download '2.0 zip'process, then the other program will skip the download '2.0 zip' process. The presupposition is you have prepare all the things it needs.
> > 
> > 
> > I just patched my Pro on my friends' Win 10 using the v2.0 and the final step showed that the firmware is 00.00.0000. Actually I though it works however the height limit still 499m I tried on the simulator within Dji Assistant 2 v.1.2.5 (official one not debug one) which didn't pass the height limitation. Because here I am not able to fly shortly probably I will try later.
> > I also tried that v1.7 offline which I didn't pass the final step of showing the NLD, and my FC showed [03.02.44.07](tel:03.02.44.07) but not the target of showing [03.02.44.10](tel:03.02.44.10).
> > Congratulations! You have already flash the firmware into the aircraft. Your aircraft can flight higher than that in the simulation.
> 
> Excuse me?

The simulator is only 500m high so you cannot fly above 500m in the simulator but it will fly higher outdoors

ywslodl, 2020-03-09:
> Sorry for you don't know my meanings. If you can't understand it, you can ignore the reply.

That’s fine not a problem. Thank you anyways

ywslodl, 2020-03-09:
> The simulator is only 500m high so you cannot fly above 500m in the simulator but it will fly higher outdoors

Thank you for the brilliant patching package. Finally I successfully patched it yesterday and yes it is true I can fly higher than 500m outdoors but limited at 500m in the simulator.

closed 2020-03-09 by ywslodl