brett8883/DJI_Super-Patcher #95: Is there anywhere I can connect my P4P v2.0 to Assistant 2 v1.1.2?

ywslodl, 2020-03-10:
I just tried to patch my friend's P4P v2.0 however I am not able to connect his drone to the Assistant 2 not v1.1.2 but the normal v1.2.5. Then, we downloaded the Assistant 2 for Phantom 4 and then it successfully opened the recognised the drone on that version. After that I also tried to reinstalled the A2 v1.1.2 however that version didn't recognise the  P4P v2.0. In this case, do I still need the A2 v1.1.2 for the parameter? Or just patched it is okay? Thank you

brett8883, 2020-03-19:
Unfortunately P4Pv2 isn’t recognized by assistant 2 1.1.2 so you can’t change parameters. Super-Patcher gets you the height limit, NFZ restrictions lifted, and FCC mode by way of jkson but if you want to change other parameters you’ll need to bake them into the firmware yourself. 

See https://github.com/o-gs/dji-firmware-tools and https://github.com/o-gs/DJI_FC_Patcher

ywslodl, 2020-03-19:
Thank you and do you mean that I can still fly over 500m height limit, fly in NFZ and enable FCC by just applying the super patcher without using assistant 2 1.1.2? Just I am not able to change the vertical speed, go home speed, max speed or similar parameters?

greektony, 2020-04-13:
hi
did you finally succeed?

greektony, 2020-04-13:
i have succefully install superpatcher2, but A2 v1.1.2 doesn't recognize my phantom4 pro v2..