brett8883/DJI_Super-Patcher #97: SuperPatcher does not work with P4ProV2

PLC-Fan, 2020-03-18:
Hi,

can anyone tell how to patch the P4PV2?

I tried by flashing SuperPatcher-downloaded 1500er stock firmware with dumdlore before, and the I porocessed SuperPatcher. But DumlDore "EnableADB" seems not working on a P4PV2. Therefore, the firmware remained 0.1500 .
You did you manage to patch?

Further question: What exactly does it mean: "Stock" firmware?

brett8883, 2020-03-19:
Sounds like you don’t have the correct ADB drivers installed. Install Assistant 2 from DJI to install them. If that doesn’t work for whatever reason some computers just don’t like to cooperate so try a different computer. I have personally used dumldore to adb to P4PV2 and it works fine.

brett8883, 2020-03-19:
Stock means original factory firmware. Not modified like Super-Patcher firmware

PLC-Fan, 2020-03-19:
Aahh, thank you, that sounds very good. :-)

 

I have installed old Assistant versions, because there was -theoretically- the possibility to change the parameters by editing an electron  Javascript file (main.js) in DJI Assistant 1.1.2. (presented by Youtube user " M1dn1ght N1nj4 ")

 

But my P42V2 didn't want to connect to 1.1.2 any more since I made my first firmware update in 2018. It doesn't connect to 1.1.2 even after downgrading to 01.00.00, now.

 

Ok, so I will install the latest DJI Assistant and tell if it works.

 

Thank you for your help.

 

Best regards,

Fabian

 

 
 

Gesendet: Donnerstag, 19. März 2020 um 04:38 Uhr

zzb1967736495, 2020-03-20:
> 听起来您没有安装正确的ADB驱动程序。通过DJI安装助手2进行安装。如果由于某种原因该方法不起作用，则某些计算机只是不喜欢合作，因此请尝试其他计算机。我个人已经使用dumldore将Adb转换为P4PV2，并且工作正常。

你好，大疆P4刷机超级固件之后，搜不到卫星信号这是咋回事，谢谢你

PLC-Fan, 2020-03-22:
Hi brett,

 

thank you for your help. :-)

 

Unfortunately, it doesn't seem to work, even with the latest DJI Assistant 2.0.8 from DJI website.

 

SupperPatcher doesn't complain, DumlDore says that ADB is enabled until next reboot, Firmware is flashed without problems, but after all, 1500 is still remaining on the system. Any idea? :-)

 

Perhpls, maybe that old drivers remain on my system despite uninstalling old Assistant versions?

 

 

Best regards from Aachen,

Fabian

 

 

P.S. On my desk, I have no wings mounted on the drone, but I think thats not the problem? :-)

 
 

Gesendet: Donnerstag, 19. März 2020 um 04:56 Uhr

brett8883, 2020-03-22:
Post your log file. Just copy and paste it here

PLC-Fan, 2020-03-22:
Hi Brett,

 

sorry, I understood, I post now via GitHub. But could you remove my name and email from your postings in GitHub?   I don't like to be too visible via google.... :-)

 

Regards,

Fabian

brett8883, 2020-03-22:
I’ve cleaned up our conversation to remove your personal information as best I can so it can’t be retrieved through GitHub’s interface. 

Please type directly into GitHub in the future. Replying via email is what causes your information to be exposed. 

Please share your log file so I can help you further.

PLC-Fan, 2020-03-22:
Ok,. :-)

brett8883, 2020-04-06:
Waiting on that log file

PLC-Fan, 2020-04-07:
Sorry! :-)

 

START SUPER_PATCHER
05.04.2020_21:49:13,54
I have verified internet access to GitHub
21:49:16,01
downloaded DUMLdore sucessfully
 
starting dummy_verify_step
 
AIRCRAFT PROPERTIES:
 
adb shell mkdir /vendor/bin
adb push dummy_verify.sh /vendor/bin/
 
dummy1 engaged based on AC type is P4Pv2 1
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /sbin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd /
adb shell mount -o remount,ro /vendor
dummy_verify step complete
Starting bind step...
bind1 engaged based on AC type is P4Pv2, verify type is 1
adb shell mount -o bind /vendor/bin/dummy_verify.sh /sbin/dji_verify
* daemon not running; starting now at tcp:5037
* daemon started successfully
Start flash step
dummy_bind complete
Start jkson
User declined FCC mod
going to _verify
_verify.cmd starting
User Indicated patch not successful. Showing log file

 

 

 

 

Gesendet: Montag, 06. April 2020 um 17:33 Uhr

brett8883, 2020-04-12:
Everything looks good to me. Have you tested it?

PLC-Fan, 2020-04-13:
Hi Brett,

 

sorry that I'm not answering via GitHub which you actually want, I guess. But pressing teh "answer" button is so simple, especially after a few good glasses of french red wine... :-)

 

I look up the firmware version via Dumldore, say always 00.1500. And the lights don't switch off although I've chosen StealthMode.

 

My alternative hope is ne new NLD version. (Current doesn't support P4P V2, yet.)

 

What makes me angry is that I'm too stupid myself to crack the system and then to help others, afterwards - altough that I'm electrical engineer with many years of software expereince. :-) But my experience is currently on automation .

 

There was the possibility to show the debug menue (and therefore the removal of height limits) in an early version of DJI Assistant (1.1.2)  by modifiying the main.js in the electron app. But my P4PV2 does not connetc anymore to 1.1.2 after upgrading the firmware 2 years ago, even if dongrading afterwards. And the later DJI Assistant versions have packed the js-files via Asar and it's at the moment impossible for me to crack the js. Therea are some videos on youtube with successful  1.1.6 crack but the autors show only that it works, not how they make it work...:-)

 

Best regards from Aachen,

Fabian

 

 

 

Gesendet: Sonntag, 12. April 2020 um 21:49 Uhr