brett8883/DJI_Super-Patcher #98: Can't fly above 500m

pointbob, 2020-03-18:
1) I must have done something wrong. I flew to 500m (1640ft) and my RC said MAX ALT REACHED and I couldnt fly higher. I successfly applied super patch just the other day. I checked to make sure the version was correct on NLD as well. Any guess as to what' s going on? i checked dji assist vers 
[image]
112 and fly limit height still at 500m  - the others are 3000m...I tried changing the 500m to 3000m but not sticking...

2) how do I know that FCC boost is actually working. I live in a city area and can barely get 1000 ft away when i start losing radio transmission (blinking radio)

pointbob, 2020-03-19:
repatched a 2nd time and it appears to have worked! thank you

brett8883, 2020-03-19:
Glad you got it to work. Closing this issue

closed 2020-03-19 by brett8883