brett8883/DJI_Super-Patcher #100: Phantom 4 Advanced, got stealth mode first time, but never again

mhaun, 2020-04-01:
I went from stock firmware 0128 on my P4A and installed stealth enabled firmware and it worked perfectly!  However, the next night I flew and no stealth mode!  I re-flashed stock firmware back on twice, and re-flashed super-patcher and all loaded with no problem, but I lost stealth mode.  I redid the procedure several times but was never able to regain stealth mode again.  Per the instructions, once you put it on then you're all set and nothing more needs to be done.  So I don't know how to correct this issue?  I hope I'm putting this concern in the right place.  Any idea how I might correct this issue?  Thank you.

brett8883, 2020-04-04:
Interesting. I may know the solution. 

It will take me some time to get to this but I will put it on my list. 

Thanks for the issue

mhaun, 2020-04-25:
Brett - I was just checking with you to see if you've been able to make any progress on this.  Thank you.

axemanarb, 2020-04-25:
All good thankyou