brett8883/DJI_Super-Patcher #101: New project help from mavic2pro

xyls13, 2020-04-01:
Hello,
I have been watching your software for a long time, and feel that software is awesome. But I do not support the DJI mavic2pro I have. I want to lift the height limit. This is the only request. Please support DJI mavic 2 Pro. Now there are more users using mavic2pro. If time and energy allow, you are sure to develop it. Thanks again.

brett8883, 2020-04-04:
That is beyond my capability at the moment. They are much much better engineers than myself working on that without much to show for it. 

Sadly, it doesn’t appear the Mavic 2 will be able to be unlocked before it is replaced.

GitHub-Updates, 2020-12-06:
Some companies are able to downgrade the anti rollback from 400+ back down to 100 to enable nfz, can you make a script and tell us the method of doing this?

Thanks