brett8883/DJI_Super-Patcher #102: internet connection to GitHub which。No way

xu-huan, 2020-04-02:
Super-Patcher will check internet connection to GitHub which。No way

brett8883, 2020-04-04:
????

The command to check internet connection to GitHub is 

`PING -n 1 140.82.113.4 |find "TTL=" && goto downloadtools`

140.82.113.4 is github’s IP address. If the ping receives a response then connection to GitHub is considered successful. If not then it fails. 

If you so wish, you can replace this line with 
`go to downloadtools`

Super-Patcher downloads the files ad hoc as to reduce the initial download size. Also, my agreement with the developer of Dumldore that I ensure the latest version of Dumldore is always used with Super-Patcher. My solution to honor this condition is to make Super-Patcher download the latest version of DUMLdore each time it is run. 

Super-Patcher cannot work without DUMLdore so it is important to me that I uphold this agreement. 

Super-Patcher is a simple batch file. It runs directly from source code that is human readable and is the easiest form of “code” to understand and is easily modified by the user. If you wish to change it yourself you are certainly welcome to and the method to do so has already been outlined by me in an earlier issue.

brett8883, 2020-04-04:
Otherwise I am not sure the nature of your issue.