brett8883/DJI_Super-Patcher #103: Battery parameters

mavicluke, 2020-04-06:
Hello,
thanks, you did a great work with your patch. Everything works but when I patched my mavic pro I choose to not apply the smart battery mod.
Is there any way I can remove the auto landing with low battery changing parameters in dji assistant?
Thank you very much.

brett8883, 2020-04-06:
Thanks for the question I have been meaning to put this information up.

Yes I made it so that you could choose later if you wanted to enable the smart battery mod features from within Assistant 2 1.1.2 in debug mode even if you do not choose the smart battery mod option when patching. Some of these parameters do not require patched firmware at all. Some of them require you to patch with Super-Patcher 2.0 or later but you can do it even if you said no to the smart battery mod. 

**The specific parameters you are looking for are:**

- Disable forced auto-landing at only enough battery to land level

`g_config_bat_config_enable_smart_bat_landing_portect=0`

`g_config_bat_config_level2_smart_battert_land=0`

`bat_level_2_action=0`

---
---

**The rest of the parameters in the smart battery modification are:**

- Set low battery warning percentage. Value is in percentage. Choose any percentage 0-100 that you would like the low battery warning to alert

`dji_bat_level_1=0`

---
 
- Set critical low battery warning percentage. Value is in percentage. Choose any percentage 0-100 that you would like the low battery warning to alert. Value must be equal to or less than <dji_bat_level_1>

`dji_bat_level_2=0`

---

- Changed calculations of only enough battery to return to the home point. Value indicates the battery percentage the Aircraft will calculate the time it needs to return to the home point and still have  `x` % battery remaining once returned to the home point.
   - I.E. Smart battery mod default value of 5 means aircraft will **arrive** at the home-point with 5% battery remaining if smart RTH is enabled in DJI Go 4. 

`smart_bat_basic_cap=5`

---

- Disabled smart RTH by default. Can be re-enabled in DJI Go 4. 

`g_config_bat_config_level1_smart_battert_gohome=0`

greeen54, 2023-02-22:
Hi!
If I run the Patcher and in smart battery mode i choose ,,Yes”, then the program overwrites the factory values or do the values ​​remain but I can change them later in Assistant 2? 
Greeting!