brett8883/DJI_Super-Patcher #104: Question Mavic PRO Drone and Controller

PipeDeveloper, 2020-04-07:
Hello comunity.

I have flashed my Mavic PRO with this firmware and it seems to be working fine. My app DJIG GO 4 shows v0000000 so it seems to be flashed correctly. Battery mod is awesome.

Question 1: Im not sure if FCC and controller boost is working fine. It seems to loose signal about 3.5km when before changing to this mod, using firmware V01.04.0500 it flights about 5km.
On the setting i have set enable FCC, signal on 2.4G (default) and enable FCC and boots controller with the objetive to reach at 6-7km. How i can test or make sure if im using the correct config? or suggestions for testing.

Quetion 2: About the controller. I always used firmware V01.04.0300 on my controller, and i never touched since i bought it. Should i update to V01.04.0500 (using oficial DJI Assistant 2)?

Thank you very much!

ikkezelf-drone, 2020-05-10:
q1:
make sure that you always flash every FW two times!

q2:
make sure that the aircraft and the RC are running the same FW version.

PipeDeveloper, 2020-05-10:
@ikkezelf-drone 
Thank you very much. For me FCC is enabled, i did some test and i may confirm with other people that have the same configuration.
its a litle weird, the distance and the signal strength seems to be the same with and without this mod. I just be sure that i flashed the firmware twice, also for signal mods.

Before closing this thread you, or anyone else according your experience flying, signal boost and FCC mode may help you to fly further?

Thank you very much