brett8883/DJI_Super-Patcher #106: P4PV2.0 can't downgrade to 1.00.1500 from 1.00.5000 this new version.

wanghuakangjq, 2020-04-12:
DJI updated the **P4PV2.0** new version to **1.00.5000**. i am try to use DUMLdore downgrade firmware to patcher needed **1.00.1500** version,but is not work , is still **1.00.5000**. what should i do it ? waiting your answer @brett8883

brett8883, 2020-04-12:
I’m looking into it. There may be a way

wanghuakangjq, 2020-04-18:
> I’m looking into it. There may be a way
Thanks for your time！

knarfboy, 2020-04-19:
1.00.5000 won't work with dumbdore anymore…..

wanghuakangjq, 2020-04-20:
> 1.00.5000 won't work with DUMLdore anymore…..

HI！boy! than how i downgrade 2.0,wait for you answer thanks!

knarfboy, 2020-04-20:
Wait - was the correct answer!

gppms, 2020-05-18:
I'm also looking for this downgrade, how stupid was from me to upgrade myP4P v2.0 to the latest firmware (1.00.50.00), i was the first guy which avoid all the time the firmware upgrades.

daftplast, 2020-05-23:
Any way to downgrade firmware from 1.00.5000. Just bought P4P v2 and it was already at 1.00.5000