brett8883/DJI_Super-Patcher #107: Platinum Cant change values in 1.1.2 after patch please help!

vidda0082, 2020-04-14:
After flashing and make sure it is. Going in debug mode but cannot change any value to gain speed wich was the point of flashing my mavic pro platinum.
Can someone help please?

brett8883, 2020-04-16:
Have you pressed enter after changing a parameter?

vidda0082, 2020-04-16:
Yes and value always reset to original. My dji4 app show blank firmware but still cant change the value. As soon as I hit enter the value resets itself to original

vidda0082, 2020-04-16:
It worked! Thanks!
But cant access to my battery settings in dji go4. Is there a way tu set it?

brett8883, 2020-04-20:
> It worked! Thanks!
> But cant access to my battery settings in dji go4. Is there a way tu set it?

Please refer to this post to adjust battery parameters when using the smart battery mod.

https://github.com/brett8883/DJI_Super-Patcher/issues/103#issuecomment-609865361