brett8883/DJI_Super-Patcher #108: Display/mobilephone/app disconnect while flying randomly.

Weirdoz, 2020-04-19:
I have a Mavic Pro Platinum with a 7 inch Cristalsky monitor with original bracket (USB-A to USB-C).
Yesterday was the 4th time i lost connection between controller and display/app (grey display, lost connection).
Restart the app and or the display has no effect, it does not connect again. Even another device as an iphone11 with different cable (micro usb/lightning) does not connect to the controller.
The connection between the controller and the aircraft ist stable, i get the bird back by push the coming home button.

Only when I restart the remote control I can connect again, the display/app to the remote.
Today i‘ve reflashed the remote firmeware, but the issue resists.
The problem is not always reproducible, it has happened 4 times so far.
Any ideas what can solve this problem?
A guy in the dji forum wrote, he also has this problem since he uses superpatcher and it still resists even if he is flashing back to standard firmware.
Any suggestions?

axemanarb, 2020-04-28:
ihad this problem today while 5 kms out , very scary , I hope when Igo back to original firmware I dont have same problem . seems this is a big problem with superpatcher . please digidat  brett HELP !!!!!!!!!

Weirdoz, 2020-04-28:
Meanwhile i am not so scared anymore, if it happens i switch off the remote and switch on again, this tooks about 10 sec. Thats the time the ac tooks to start "coming home" process.
If you have used fcc or boost before, i‘ll do it by switch, it will be back in that mode like before.
It has always worked, just the first 2 or 3 times it was a little bit scary.
I am not sure if superpatcher was the reason, the last 5 flights it does not appear again.

axemanarb, 2020-04-28:
I did switch off remote and back on and the app did not re connect . I might try it a few more times and see how it goes . It must be from the patcher , everything worked fine until I did it . Another fellow in here had his drone fly away and crash so it is definatly  a big concern .

Weirdoz, 2020-04-28:
Fly aways are not only a problem with or of the superpatcher. It happens anyway very often with drones.
My problem was 100% always resolved after the remot restart.

axemanarb, 2020-04-28:
I will try again tomorrow and see how I go . This time I might not fly 5
kms away .

axemanarb, 2020-04-28:
Any problems with your speed or with switching between sport and gps mode .
I could only get max speed 32kms ph and when in sport mode it was same as
gps mode

Weirdoz, 2020-04-28:
nope, can easly switch between atandard and sport mode. Drone is about 80km/h fast with no wind in sport and 30 km in standard mode.

axemanarb, 2020-04-28:
I got the speed back after I used litchi but it doesnt drift in sport mode
. Very strange but I will do more tests soon . I just want everything to
work . Ahhhhh I will get there one day .

ikkezelf-drone, 2020-05-09:
> Any problems with your speed or with switching between sport and gps mode . I could only get max speed 32kms ph and when in sport mode it was same as gps mode
> […](#)

Same here,
I started the nolimitdronez client and saw that the sport mode settings were almost the same as the normal mode.
You need to change them (back) to your desired values. I did that with the nolimitdrones client as I can't seem to get that dji assistant in debug mode

ikkezelf-drone, 2020-05-09:
I installed the stock firmware (downloaded by the super-patcher) and after that I flashed it to the 0500 fw to get rid of the update messages.
Now I'm facing also the lost video feed/aircraft disconnect issues. using a platinum mavic.

I then started the same procedure again but now without the 0500 flash.
I got to 750 meter above ground in the sport mode (which was not using any sport speed settings).
when I returned to the ground I got signal los again.

now after reflash to the SP version I used NLD to put back the sport speed settings. and will try again.
keep you poosted.

ikkezelf-drone, 2020-05-10:
my observation was that the RC still works but the connection between phone and RC/aircraft is gone.
I couldn't get it back up until I shutdown the drone and started it back up again.

It appears that following the SP guide again AND applying the 0300 -> mod-by-brett-0500 flash TWICE! did solve the issue for now (now tested it two battery life minutes and no disconnect sofar)

Weirdoz, 2020-05-10:
I have done every flash 2 times, but the issue still happend again.
It is so randomly that i can fly 3 to 6 times without any issue and the it appears again.
Andt it has no reason in interferences or something like it, because the testflights where always out in nowhere, and always the same place.

Weirdoz, 2020-05-10:
> my observation was that the RC still works but the connection between phone and RC/aircraft is gone.
> I couldn't get it back up until I shutdown the drone and started it back up again.
>
> It appears [...]

have you ever tried to restart just the remote, you can do it right when the bird is flying. Just be sure you have the right height for RTH configurated. You can restart the remote as fast as the bird starts rth, and then stop it. It never happend again in the same flight after restart the remote.

booooooooooooh, 2025-02-03:
Hi Carlie could you please hold off on taking it out of my sick leave till
i see the doctor please.
Thanks