brett8883/DJI_Super-Patcher #109: P4P aircraft lost 😭

MatthewPoland, 2020-04-19:
Hello,
after installing patcher for only NFL the aircraft disconnects after distance 300m and 270m height 9min in vision of sight and gone! I don’t know if it's problem with patcher in European GPS coordinates or what? The DJI app log is showing correct home point and good GPS signal. Only couple of interference errors. 
So sad for my lost. 
And angry on myself.
This flight wasn't in NFZ.
I was flying in same spot earlier this day with Mavic 2 zoom with so issue at all. Switch to P4p for better low light performance 😢
I can provide log file if anyone can help with understanding this. For now I removed all mods from all my other aircrafts.
https://app.airdata.com/share/lIAweH

axemanarb, 2020-04-28:
so sorry for your loss , I to had a very scarey moment today after doing superpatcher I had a disconnect and never re connected but luckily nmy rth worked . sos sorry mate . I will be uninstalling superpatcher until someone can tell me it is safe but as it stands it doesnt look good

MatthewPoland, 2020-04-28:
Hi, 
after two days of searching I find a guy in my town who post on local site that he found a drone. Luckly it turns out that was my drone :D He found it 100m away from last point form DJI GO App crashed on the ground, battery was poped out but still 60% of charge, two propellers broken couple of plastic parts as well. When I was trying to check the logs inside the drone there was same log that inside DJI GO App and after that couple of corupted files without any convinienc data. No battery or motor status, no altitude logs etc. Probably the drone just disconnect and turn off, then after 260m of falling hit the ground, maybe bounced from the tree that could lower the impact a bit. But thats it. I send the logs on DJI forum but without the answer, sorry they said to send the drone to their facility. Without the warranty and custom firmware probably they will charge three figure amount XXX$ and just return it with oryginal firmware blame on it.

I will repair it and put the original firmware, do a couple of test of the battery and see how it works. But for now I'm a little bit scared to fly this drone.

Cheers!

axemanarb, 2020-04-28:
So happy to hear you got it back and you can fix it . Questions need to be asked and answers had so we can feel safe in modding our birds . 
All the best my friend .

axemanarb, 2020-04-28:
I would like to see the results of the log files

MatthewPoland, 2020-04-28:
Here you go!
https://app.airdata.com/share/oaDoum
https://app.airdata.com/share/lIAweH

axemanarb, 2020-04-28:
Thanks mate so what was the problem ?

MatthewPoland, 2020-04-28:
I don't know. It's not clear what happend. No data in logs after disconnect. Only some scraps of files with flat data :(

axemanarb, 2020-04-28:
Hmm probably DJI worming their way in some how . Sooo sneaky.

MatthewPoland, 2020-04-28:
probably I will not send it to DJI to pay extra $ for a crappy answer, and it will last 4 months as I read on forum with similar cases.

axemanarb, 2020-04-28:
Dji will see whats been going on and void the warranty and for sure charge
you big $$$