brett8883/DJI_Super-Patcher #111: superpatcher. mavic  disconnected and would not re connect on /rth

axemanarb, 2020-04-28:
totally disconnected  when i was out pretty far / rth kicked in but did not reconnect when in close . also my speed was only 30 kms/ph  and sport mode was the same as gps mode as in the drone was fully stable in one place in sport mode , no drift in wind . I changed from dji go 4 to litchi same problem with sport and gps . at first my speed was 34 kms / ph max then kicked into 64 kms/ph . I have not checked how it is now with dji go 4 yet. anyone else have these problems ? I was pretty scared I would lose my drone on disconnect

Weirdoz, 2020-04-28:
So i saw you find my thread about it, just switch off and on thze remote and you get connection with the app again. The remote is still connected to the ac, right, even when the phone/app breaks?

axemanarb, 2020-04-28:
Yes the remote was still working thank god as I could see how far away it was and it was in fact returning to home . I tried turning it off and on but still no re connection with the app . It has me very worried to use superpatcher and we are not the only ones so I hope Brett / digidat aka.... can fix it or help .

axemanarb, 2020-04-28:
I also tried turni g the app off and on and still no good . I dont know if I should try again tomorrow using the litchi app

Weirdoz, 2020-04-28:
Did you switch off the remote or the app/phone? There is a problem with the USB Ports (both), if you restart your remote, all works fine again and will not appear again for the next few flights.
It has notghing todo with the app or phone, its a problem with the remote.

axemanarb, 2020-04-28:
I switched off and on the remote . I am using the usb potd underneath to usb c cable to my 2019 samsung tab a 10.1 inch . fast great tablet.

axemanarb, 2020-04-28:
@brett8883 any ideas ?

axemanarb, 2020-05-03:
Sorry Edit a bit Im going from a p3a to mavic totally new to how it works . Sport is not atti mode . Got it . Im looking at changing it to atti mode . I like how it dont brake so hard and can film in atti smoother .

brett8883, 2020-05-08:
You were probably just in a high interference area. Signal disconnects happen. It returned to home which is what it’s designed to do. 

This doesn’t sound like a Super-Patcher issue.

Weirdoz, 2020-05-08:
No it is not a signal disconnect, it looks like the USB Ports are not working until we restart the remote control. The signal from the aircraft to the remote is stable, just the connection from the remote to the crystalsky/mobile device via USB (both ports) are not working.

ikkezelf-drone, 2020-05-10:
my observation was that the RC still works but the connection between phone and RC/aircraft is gone.
I couldn't get it back up until I shutdown the drone and started it back up again.

It appears that following the SP guide again AND applying the 0300 -> mod-by-brett-0500 flash TWICE! did solve the issue for now (how tested it two battery life minutes and no disconnect sofar)

booooooooooooh, 2020-05-15:
Thanks . I need to do much more reading .

brett8883, 2020-05-18:
Super-Patcher doesn’t do anything to the RC. This issue is unrelated to Super-Patcher

dongfeng219, 2020-05-28:
。。。。。。。。