brett8883/DJI_Super-Patcher #112: P4PV2 succefully patched with fw2200 bin file, but cant remove auto landing..

greektony, 2020-04-28:
Hallo github,
I have successfully patched my P4PV2(v2200).
I have try it with both bin files (2200 & 1500).. When I use the 2200bin file I cant disable auto landing. In 1500bin file auto landing seems to be disabled, but I have no gps signal at all..
In both cases I have enable smart battery modification when superpatcher asks during installation process.
So, Both cases have a flaw.. of course I prefer the 2200bin case because I can fly my P4PV2, but I cant go high enough because auto landing kicks in and force me to land.. 
Has anybody similar issue?
What can I do?

brett8883, 2020-05-02:
Here’s a .1500 flight controller without Galileo but has the battery mod. 

https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/P4Pv2/2.1/P4Pv2_Batt-Mod_NoGal_03.03.04.56_dji_system.bin

I don’t have one made for .2200 that has the smart battery mod yet.

greektony, 2020-05-02:
Thank you Brett, it works great..

closed 2020-05-05 by greektony