brett8883/DJI_Super-Patcher #113: I executed Super Patcher on Mavic Pro (FW v01.03.0700) now I can't connect with DUMLdore

uLightMe, 2020-04-29:
When plugged into the computer the MP seems to be stuck in a boot loop constantly showing connected/disconnected in device manager (Unknown Device)

I know I screwed up by running Super Patcher on the wrong firmware, but is there anything I can do now to get my Mavic working?

Thank you.

axemanarb, 2020-04-29:
Try use dumbldore and flash the old firmware back a few times ? Maybe ?

uLightMe, 2020-04-29:
I can't connect to the PC with a USB cable because it (connects and disconnects) constantly.
Dumbldore doesn't recognize the drone.  
The drone shows up as "Unknown Device" in device manager.

axemanarb, 2020-04-29:
I know I flashed my p3a with an sd card im not sure if the mavic is the same . Does it show what firmware your on in your app ? Btw im a total newbie so dont take my advise im sure someone will help you fix it @brett8883  ???

axemanarb, 2020-04-29:
Wait till you get the pro advise first before do anything

uLightMe, 2020-04-29:
> 
> 
> Wait till you get the pro advise first before do anything

I appreciate you trying to help.  I hope someone here can help me.

axemanarb, 2020-04-29:
Does it connect to the app and show what firmware your on?

axemanarb, 2020-04-29:
http://dankdronedownloader.com/DDD2/app/index.php?cat=Mavic%20Pro%201%20-%20Incl%20Platinum%20and%20Alpine

uLightMe, 2020-04-29:
No. 
I recorded a video of what's happening when I connect it to the computer.
https://youtu.be/TkJi6n9KYxg

uLightMe, 2020-04-29:
it keeps doing the connect/disconnect (Unknown Device) constantly without an end.

uLightMe, 2020-04-29:
GOOD NEWS!
After restarting my Mavic for 5 or 6th time and letting it sit there doing what you see in the video for a while.  It now connects to the computer fine and I can flash the correct firmware :) whooohooo.

It would seem like whatever I did before got it stuck in this repeating boot loop during which it was "healing itself" LOL .... finally it turned on and I can connect with DUMLdore again.

closed 2020-04-29 by uLightMe