brett8883/DJI_Super-Patcher #115: failure of imu and mavic pro is loss

hardnv, 2020-05-10:
Hi!
today during the second flight I lost control and the signal of my mavic pro, immediately after takeoff the control of the drone and a complete orientation failure and the next loss of signal failed.  I have a main question, why did the program let the drone take off before imu initialization or is there a problem with the sensors ??  I think the problem is in this patch.  since in stock he does not allow this!!
https://app.airdata.com/share/sxIFTk

PipeDeveloper, 2020-05-11:
@hardnv im sorry to hear that you loose your drone. Did you recover it? I hope that you recover its parts.

As i know the firmware always do the checklist, battery, sensors, etc, Always they will warn you. The thing here is that the take off for imu is disabled, this is because you may turn-on motors even if your drone is facing up. This is for example, you may turn off motors in the air, and then turn them on again, on take off drone will fall more faster but not loosing the horizontal stability, then you may increase power to fly normal, specially for high altitudes.

**Always before take off (with or without the mood) is important to check and calibrate your sensors, imu and compass.** Even if your imu is not calibrated (for example with yellow warning and high values) the original firmware also let you take off, but here the worst thing you make experience is that your drone will go directly to one way (thinking its horizontal) and may crash if you dont manually control it faster.

I see your flight log and seems you got several errors. But i dont see that the imu was the problem. Remember that he imu is a a sensor, located under de ESC board, this like an osciloscope, it measure the XYZ axis in real time, that means that knows the horizontal position for stability, measure velocity and calculate angles for moving, break just by gravity.

The main thing is a hardware problem on the ESC board of your drone (watching your flight log). This board is responsible to distrute energy to all components on your drone, camera, motors, sensors, GPS, main board, etc. This is just a chain issue, if this fail, all components will fail. That's why you experienced lost of signal and sensors failure + no control for compass failure (important for face direction) that means your drone probably rotates in itself, and GPS for keep on the same position. As i said it is just a chain issue.

There are many ways that could be happen, for example, a wrong contact between your battery and the pinouts, oxidation, who knows, You need to meditate

axemanarb, 2020-05-12:
I to did superpatcher on mavic platinum.  I then changed a few things on DJI assistant.  I changed it to atti mode then didnt like it and changed it back . I had a fky after and no problem other than I misjudged the return anount of battery . Anyway after i was inside my house and i was  in gps mode then suddenly it changed into atti mode now bare in mind I had changed it back so it couldnt go into atti mode anymore in assistant . Anyway once it strangely changed from gps to atti mode by itself I crashd into the wall, the drone revving off its head I finaly stopped it and my props and gimbal is damaged . This is not the first problem I have encountered since superpatcher but this problem eas a very dangerous one . What if it flew into some kids revving so high . Another problem is there is a speed value I change from 10 to 20 on assistant and it will not stick when other values I change do .
I am hoping there is not too much damage to my new platinum and I am also really hoping I can get rid of superpatcher without any problems afterwards with my drone . I wish now I had of just made the changes myself with assistant and hope that works without any drama . If anyone has had similar experiences please share as peoples safety is paramount and I also dont want to lose , wreck or fork out more money on my on my drone hobby. If others can shed light on this subject it helps me any many others . Please if you can help . No offence to the maker of superpatcher if this is a culprit in any experiences as im sure his hard work and his heart is in the right place . If there is any warranted feedback im sure it is accepted.  As for me I am a newbie and will never get to a level such as people like Brett digi . All this said I tgink this is a top priority subject as it is a safety issue so all help is appreciated.

axemanarb, 2020-05-12:
I also need to say being a newbie that superpatcher may not have caused this problem.  Any help getting to the bottom of this much appreciated

hardnv, 2020-05-13:
@PipeDeveloper The drone was not found, it was finally lost.
This is not the first complaint about the Super-Patcher, in our community a message of half a year or more has a lot of complaints about the drone’s crush and loss after applying the Super-Patcher

axemanarb, 2020-05-13:
I will be uninstalling superpatcher.

brett8883, 2020-05-13:
Flying a drone indoors will cause a loss of GPS signal and will cause it to go into ATTI mode. This is normal and expected behavior with or without the mod. 

The motors not turning off automatically after a crash is documented and is an intended modification. It's meant to allow the drone to try to save itself instead of giving up and falling out of the sky. 

Don't fly the drone into children and you won't have an issue. If you think that is a realistic possibility this modification is not right for you.

axemanarb, 2020-05-13:
Hi looks like I need to read the manual . Sorry I can see why you would need to reply and sort that out . Once again my apologies   for taking up more of your time on a simple thing I should have known . I came from flying a p3a to the mavic and thought there was only gps and sport mode . Of course there would be no gps signal inside causing  the crash . Im  not the sharpest tool in the shed . I have trouble with my memory which gets me inti trouble . Well I hope me putting my foot in my mouth might be of some help in some way to someone and if not then feel free to delete my comments. It certainly has taught me something about flying near people as you are so right about flying near people . If my drone , under my control was to hurt anyone for whatever reason , it would firstly be very upsetting for the person hurtand their loved ones . Something thst would upset me very much , knowing "I"am responsible  and also a stain on the hobby  I love . That being said it looks like before the next time I fly I know everything I need to know . 
Big lesson right here for me and once again apologies to you and I do remember seeing you say at your own risk and fly responsibly  even if there was a problem with the superpatcher.  I will be taking my time fixing my mavic to go right back over all the basics of drone flying before I attempt anymore " modding". 

Scampering off into cyberspace with my tail between my legs.......

axemanarb, 2020-05-13:
Another stupid question I should know and or look up myself .... should the downward sensors kick in if gps signal is lost ?

PipeDeveloper, 2020-05-13:
By firmware if your drone is in the air and there is no signal. After 15seconds it will automatically return to home.
For example you are 2km far away and someone uses a jammer, blocking your RC signal and GPS (most extreme case). On that case your drone will use the compass, as it doesnt have GPS in the middle of the space, it knows where you are by the compass, so it will fly to return to home hoping that in the middle of the track it will have signal again. Not too much precise +/- 5 degrees but enought to fly in the right direction.

brett8883, 2020-05-14:
> Another stupid question I should know and or look up myself .... should the downward sensors kick in if gps signal is lost ?

The downward sensors should provide positioning   given the conditions for VPS are met. VPS is enabled, the aircraft is close enough to the ground, enough ambient light, and a distinct and recognizable pattern below. If these conditions are met and the drone cannot receive GPS signal the drone will switch to OPTI mode. 

It usually switches to ATTI mode momentarily before switching to OPTI mode in my experience so you have to be on your toes. Most people who fly indoors prefer to force the drone to switch to ATTI mode because then they know it won’t switch out of that mode. Even though it is more difficult to maneuver it is seen as an improvement over the drone switching modes without warning. Atti mode is easy once you know what you are doing.

axemanarb, 2020-05-15:
Thanks again .Appreciate your responding .

gppms, 2020-05-18:
I also had problems with my MP1 when i installed the super patcher, the mavic was unstable while hovering and one time unlock from the gps mode and kick in the attitude mode and start to drifting with the reception of 15 satellites. finally i uninstall the super patcher and made all the parameters modification with the assistant. I personally believe that the super patcher is dangerous.