brett8883/DJI_Super-Patcher #117: incompatibility with Mavic Pro Platinum?

Theo740, 2020-05-16:
Superpatcher works great with Mavic Pro, however, it seems to cause problems with the Platinum. I had Problems with a sometimes crashing Platinum, couldn't find the reason. Then I updated a Pro with a Platinum ESC and that Pro then also started crashing in the same way. Mostly at high sink rates plus forward or other movement in 'supersport' mode. No crashes in normal mode. I remounted the original PRO ESC and the crashes never happened again. I got myself another Pro ESC board which I will install in my Platinum.

booooooooooooh, 2020-05-16:
Interesting

brett8883, 2020-05-18:
I have a Platinum and it works fine.

Theo740, 2020-05-19:
So maybe it's due to the parameters?
FW is 01.04.0300
Parameters:
g_config_mode_sport_cfg_tilt_atti_range | Maximum tilt(tilt=speed) | 40
g_config_mode_sport_cfg_vert_vel_up | Max speed you gain altitude | 8
g_config_mode_sport_cfg_vert_vel_down | Max speed you lose altitude | -10
g_config_mode_sport_cfg_vert_acc_up | Max speed you accelerate in gaining altitude | 8
g_config_mode_sport_cfg_vert_acc_down | Max speed you accelerate in losing altitude | -10

Theo740, 2020-05-19:
By the way: Thanks brett8883 for the great work!

brett8883, 2020-05-19:
> So maybe it's due to the parameters?
> FW is 01.04.0300
> Parameters:
> g_config_mode_sport_cfg_tilt_atti_range | Maximum tilt(tilt=speed) | 40
> g_config_mode_sport_cfg_vert_vel_up | Max speed you gain [...]

What do you mean by “crashes“? Like you flew into something and crashed it?

Theo740, 2020-05-20:
It just fell out of the sky in a quick rotation and hit the ground.

Theo740, 2020-05-20:
Meanwhile I changed the ESC in the Platinum and did some stress tests while flying, seems to work fine.

brett8883, 2020-05-22:
So it didn’t recover from the decent in time? Even if it flipped over it should right itself quickly. Maybe the different props? 

No we tried that, even at -20 decent speed it was the same speed as -10. I imagine there’s a minimum RPM parameter but what else you would affect by changing this idk.