brett8883/DJI_Super-Patcher #120: major issue with i2 please please help

giannhsath, 2020-05-25:
hi brett.
after sp installation on my i2 i have a big issue with gps. i2 losing sats during flight and go to atti mode. i checked that happened when loads sats from gallileo when gets signal from more than 23 sats gps crash and the aircraft is entering in atti mode. now i try to rollback in stock fw and it was successful but  the list of sats it remain inside the aircraft i try several updates with many stock fw but this sat list is still in my i2 and keep doing the same thing...none of the updates replace the sats list with the default one...so my i2 with any stock fw inside it keeps the sat list with galileo that added with sp..... please help this i2 to fly again... 
i think i need a new sp file with the default satlist  inside...please help....

brett8883, 2020-05-25:
> hi brett.
> after sp installation on my i2 i have a big issue with gps. i2 losing sats during flight and go to atti mode. i checked that happened when loads sats from gallileo when gets signal from [...]

Sorry for the trouble. Haven’t heard of I2 having this issue. In the next version of Super-Patcher Galileo will be an optional modification. I already have the bin files made so here is the bin file with stock GPS/GLONASS settings

https://github.com/brett8883/Super-Firmware_Cache/raw/2.0/I2/2.1/I2_Standard_NoGal_03.03.66.66_dji_system.bin

This one doesn’t have the battery mod nor stealth mod but if you want one or both of those lmk as I do have those made. 

You’ll need to be on the correct firmware for super-patcher and do the Super-Patcher process but install this .bin file instead of the one Super-Patcher tells you to. 

If you want to flash back to stock after you can but you may not want to since you’ll have the mods you originally wanted.

giannhsath, 2020-05-26:
hi brett
thanx for your quick response
i flashed today the .bin you sent me and everything seems to be normal again.
thanx again.

Dronesafe, 2020-06-16:
> hi brett
> thanx for your quick response
> i flashed today the .bin you sent me and everything seems to be normal again.
> thanx again.

Hi giannhsath:
May I ask how did you solve it? I got the same issue, and flashed the .bin 03.03.66.66, but I2 was still in atti mode losing GPS. Hope for your help.

giannhsath, 2020-06-16:
yes everything seems to be ok now. just flash the .bin that brett send to me above instead the other one and make sure you are in the correct stock fw.

Dronesafe, 2020-06-16:
> yes everything seems to be ok now. just flash the .bin that brett send to me above instead the other one and make sure you are in the correct stock fw.

Shoud I flash back to the stock fw v01.02.0200，than do the patch？

giannhsath, 2020-06-16:
i suppose yes