brett8883/DJI_Super-Patcher #126: Super-Patcher will not connect to the internet.

koreandude79, 2020-06-19:
Need help.  Super-Patcher will not run connect to the internet.  I open it up and the screen opens up and closes right away.  Im running Windows 10 PC.

brett8883, 2020-06-23:
Make sure to extract the repo from the zip archive first. There’s an error message if not connected to the internet so that’s not the issue

koreandude79, 2020-06-23:
Hi bret8883, how do I extract the repo from the zip archive?  I saved the file to a folder and use Winrar to extract the file and I run super-patcher.  It starts the download process and the screen closes right away.

brett8883, 2020-06-29:
> Hi bret8883, how do I extract the repo from the zip archive? I saved the file to a folder and use Winrar to extract the file and I run super-patcher. It starts the download process and the screen closes right away.

That’s fine but you need to extract all the contents of the repo to that folder then. You are probably missing busybox which is the likely issue.