brett8883/DJI_Super-Patcher #128: Cannot find the file <MavicPro_SP_2.0_Standard__dji_system.bin> in  <SP_Flight_Controllers folder> with win10

prince2100, 2020-06-30:
HI there! Thanks for your sharing!
I have met some problems.
I cannot find the file <MavicPro_SP_2.0_Standard__dji_system.bin> in folder  <SP_Flight_Controllers folder> with win10!
same situation happens when I find firmware.bin in the folder STOCK_FIRMWARE ! However, I downloaded it from other link you provided in another page. 
So, I cannot finish this mission without necessary xxx.bin, and could you please share it in a new link?  : D  plzzzzzz!!
Thank you for reading my questions !

rlsolomon, 2020-06-30:
Hi Prince2100, this is probably the same as issue #122 so either make the code change I suggested or just pick the file which is actually created in that directory.

prince2100, 2020-06-30:
Thanks for your replying.
I mean there is empty in Flght_Controllers folder.
there is nothing in it.  I believe it’s not the problem with network, as when I change to personal hotspot the same issue happens. what should I do?