brett8883/DJI_Super-Patcher #129: Dumboedor quits while flashing

gqbigpaps, 2020-06-30:
While flashing the generated .bin dumbdore gose beyond 100percent then closes. The spark keeps beeping and reboots twice. Ive waited a while assuming while dumbdore crashed the upload was dine to the spark, so maybe it will finishing, however since dumbdore closes the only way I cam tell it is safe to revoot the spark was to wait a good 10 min untile the beeping amd rebooting stopped. 
Any suggestion??

diegobarracuda, 2020-07-18:
Hello. Exactly the same Is happening to me with my spark. I ve tried several times, but always the same when flashing the patch firmware (with stock firmware everything runs ok).

xtino168, 2022-07-23:
Yes happened to me too

mine2209, 2023-05-29:
I have the same happening to me everytime it disappears during flash it works sometimes and other times I have to run it again, find the same with Jkson mod takes a few goes to work and someties needs to run through the who patch again from stratch