brett8883/DJI_Super-Patcher #130: I2 GPS issue after SP

Dronesafe, 2020-07-06:
hi brett.
After sp installation(with patch I2_Standard_YesGal_03.03.22.22_dji_system.bin) on my I2, it's GPS always losing sats and go to Atti mode. Than I tried the patch I2_Standard_NoGal_03.03.66.66_dji_system.bin, the issue was till there. When I flashed the stock FW back to V01.02.0100, the GPS became normal. But as soon as I updated the stock FW to V01.02.0200, the issue occured again. It seems that the patch I2_Standard_NoGal_03.03.66.66_dji_system.bin can't make the GPS to stock settings. How can I do with it, could you help me?
Thanks!

hawkeyemedia, 2020-08-07:
I have the same problem with my I2. Tried downgrading to V01.01.0200, problem solved, only show an average of 15 satellites. As soon as I upgrade back to stock V01.02.0200, and no patch, it goes back to showing 23 or more satellites. Then it will intermittently show only one or two bars (low dop) and switch to ATTI, even though it shows 26 or 27 satellites. Then it will drop the GPS signal altogether. I've tried patching with the I2_Standard_NoGal_03.03.66.66_dji_system.bin patch, but it doesn't disable the Galileo satellites.

Pathogenius, 2020-09-05:
Confirmed, M1P switches to ATTI mode for no apparent reason and mode switch will not restore.