brett8883/DJI_Super-Patcher #131: DUMLdore download location

jezzab, 2020-07-07:
Is DUMLdore downloaded from my repo on the first run or is it included in the SuperTools archive? I see it is here:
https://github.com/brett8883/Super-Tools/blob/master/TOOLS/DUMLdoreV3.exe

brett8883, 2020-07-26:
> Is DUMLdore downloaded from my repo on the first run or is it included in the SuperTools archive? I see it is here:
> https://github.com/brett8883/Super-Tools/blob/master/TOOLS/DUMLdoreV3.exe

Admittedly my Super-Tools repo is a bit of an organized mess. As to not break the backwards functionality of earlier already downloaded versions of Super-Patcher I created new branches in Super-Tools for each new version of Super-Patcher. 

The current version of Super-Tools being used by Super-Patcher 2.0.2 is located here https://github.com/brett8883/Super-Tools/tree/2.0.2

The master branch of Super-Tools corresponds to only some of the earliest versions of Super-Patcher. I can’t remember which version makes the change to downloading dumldore from your repo at each runtime but it corresponds to when I told you I had made that change. 

You’ll see that in the most recent (2.0.2) version of SP, for example, there is no DUMLdore.exe in that branch of Super-tools and it does get downloaded directly from your repo at each run time and any existing versions held over from a previous run is overwritten ensuring that only the latest version of DUMLdore downloaded from your master branch is used each and every time sp is run.

Let me know if you have any questions and thank you on behalf of everyone in the community for your hard work and gorgeous program

vicmanhg, 2020-11-21:
can i downgrade the 0620 frimware here?