brett8883/DJI_Super-Patcher #143: Use DJI Spark with SuperPatcher

Consti10, 2020-09-30:
Hello,
I am opening this issue such that other users don't make the same mistake:
When using the DJI SuperPatcher v2.0.2 to flash your spark you have to be carefull selecting the right image.

In the first step (flashing the 1.0900 original firmware ) you are prompted to the right location in Dumbledore.
In the second step (flashing the modified firmware) you have to manually select the right file - not the one from step 01 again.

Consti10, 2020-09-30:
Also, when using the fcc mod make sure to first load the superpatcher firmware , then do the fcc mod.
Doing fcc mod on original firmware won't work.

jakeostrac, 2021-01-10:
are you expriencing any GPS disconnection issues after using the super patcher?

Consti10, 2021-01-10:
not at all

xtino168, 2022-07-23:
I got csc error on airdata, i notice yaw error on height limit