brett8883/DJI_Super-Patcher #146: P4P2 Not Modding

wearechangehartford, 2020-10-11:
I think I might be having an issue with DUMBLdore engaging the ADB because the front lights on the P4P2 do not no matter how long I wait (a few mintutes) and the program tells me that ADB is complete

also when flashing the bin file in /SP_Flight_Controllers (which does not match the name given in the instructions) the P4P2 does not turn itself off and on (reboot) like stated in the instructions.

I am using win 10 (which the only other topic about "no devices/emulator found" was about, they were using Win7/8 which was the issue)

when i connect to the RC and check the version, it's still V01.00.1500 and now the camera does not work (black screen when in DJIGO). i reflash the stock firmware and the drone works like normal again.

any help would be great. thanks :)

the log file

START SUPER_PATCHER 
Sun 10/11/2020_17:41:16.82 
I have verified internet access to GitHub 
17:41:18.99 
downloaded DUMLdore sucessfully 
 
starting dummy_verify_step 
 
AIRCRAFT PROPERTIES: 
 
error: no devices/emulators found
adb shell mkdir /vendor/bin 
error: no devices/emulators found
adb push dummy_verify.sh /vendor/bin/ 
 
dummy1 engaged based on AC type is P4Pv2 1 
adb shell cd /vendor/bin/; chown root:root dummy_verify.sh; chmod 755 dummy_verify.sh; cp /sbin/dji_verify /vendor/bin/original_dji_verify_copy; sync; cd / 
error: no devices/emulators found
adb shell mount -o remount,ro /vendor 
error: no devices/emulators found
dummy_verify step complete 
Starting bind step... 
bind1 engaged based on AC type is P4Pv2, verify type is 1 
adb shell mount -o bind /vendor/bin/dummy_verify.sh /sbin/dji_verify 
* daemon not running; starting now at tcp:5037
* daemon started successfully
error: no devices/emulators found
Start flash step 
dummy_bind complete 
Start jkson 
User requested info on jkson 
User declined FCC mod 
going to _verify 
_verify.cmd starting 
User Indicated patch not successful. Showing log file

schimmi6, 2020-10-19:
Hi @wearechangehartford 

how did you made the downgrade of the FW? Or didn't you have to do that?
I am currently stuck in V01.00.5000. 

DUMLdore is failing to downgrade the firmware
I can also confirm that DUMLdore is not making any changes to my p4p2.

wearechangehartford, 2020-10-20:
Yeah I'm an idiot. I dont think the super patcher can mod the P4P V2 

Like you I'm stuck with the same firmware. I think DJI encrypted the software and people are having trouble cracking it? I guess we'll just ahve to cross our fingers and have to wait until someone does. Damn those red commies!! lol

Even the NLD (no limit dronez) software isn't yet capable of modding the V2.