brett8883/DJI_Super-Patcher #148: Phantom 4 Pro GPS issue

lis-uav, 2020-11-28:
I'm getting a repeatable error on my Phantom 4 Pro V1 using Superpatcher.  It is dropping the GPS intermittently for no more than a second at a time, randomly in flight.   When I put the firmware back to stock DJI, the issue goes away.  Put Super patcher back on, and it comes back on.

It's been suggested that it may be something to do with the Galileo satellite reception being enabled.  Can this be set to optional?  Or can it be disabled when installing the Superpatcher flight controller?  From what I can tell it is not coded as a parameter?

Cheers.

giorgos66, 2020-12-04:
Strange that there is not an answer from its creators Superpatcher 2.0.2  I wanted to install it but with what I read, I am afraid.

norfeas, 2020-12-04:
> 
> 
> Strange that there is not an answer from its creators Superpatcher 2.0.2 I wanted to install it but with what I read, I am afraid.

Γιώργο καλησπέρα. Μη βάζεις το Superpatcher στο Phantom το ποιο πιθανό να πάθει ζημιά χοντρή...η ποιο σίγουρη μέθοδος είναι το DroneHacks  με το CustomFC https://drone-hacks.com/birdmap . Είναι με πληρωμή αλλά είναι σίγουρη δουλειά και έχουν και καλή υποστήριξη αν κάτι πάει στραβά. Εδώ δεν θα βρεις άκρη ο τύπος το έχει παρατήσει το souperpatcher. Φιλικά Νίκος.

giorgos66, 2020-12-04:
Καλησπέρα φίλε Νίκο. Ναι αυτο κατάλαβα και για αυτο το εγραψα.... Δεν ειναι να παίζει κανείς με αυτα τα πράγματα. Ευχαριστώ θα δω το αλλο που μου προτείνεις και βλέπω. Βασικά για καλύτερο σήμα έλεγα.

lis-uav, 2020-12-04:
After asking the question,. I did a fair bit of reading through the closed issues, and other people issues.  It seems like there is an issue with the P4Pro V1 GPS, with Galileo satellite reception enabled.  The issue appears to manifest when greater than 26 satellites are being used, and the issue is the GPS signal dropping out completely intermittently.
 
It's referenced here:  #17  

The solution is flashing the GPS controller (Module? Not sure on the terminology here sorry) back to stock, so as not to use the Galileo option.  There is a method given in the issue above, but I have not had chance to test it out yet.  Fingers crossed for a few hours spare this weekend to give it a go.

As for Drone Hacks, I'm sure that is a capable bit of software.  If there is an open source method of acheiving the same end result though, I'll give that a go.  I'm not sure why, but I have a thing about spending (even small amounts) of money on stuff like this.  Even the drone was £1200 plus!  Yep, it's illogical, but that's me!

I'll report back after trying to put the GPS back to stock.

giorgos66, 2020-12-04:
> What a pretty cheeky and arrogant way to ask questions, Brett8883 does it all for free for you. If you don't like it or if you are too incapable to handle the problems with your computers, go to other portals that charge the same money. Otherwise just wait for brett to contact you, he doesn't sit in front of his computer all the time and waits for an idiot to come and ask a question.
> So far he could answer almost all questions and i'm sure he could answer yours as well, maybe you should look in the closed inquiries.

In other words, has the problem with Super-Patcher v2.0.2 and the satellites been solved? We have machines that cost a lot of money so we want to make sure it works properly.

Weirdoz, 2020-12-05:
In other words, there always remains a risk when you flash the firmware on a device, no matter how expensive or cheap. To expect 24/7 support from a vendor who offered a method to open many possibilities for free long before anybody else is just cheeky and disrespectful. Especially since the commercial providers often do not answer at all.

lis-uav, 2020-12-05:
Hi folks.  Just an update.  It's been chucking it down with rain here for most of the day, so I've managed to get a version of firmware flashed that has the Galileo sat reception disabled (from here: https://github.com/brett8883/DJI_Super-Patcher/issues/17#issuecomment-536314530).

 I had to do the usual twice flash of stock DJI firmware, before putting this one on.  It's sat outside in the garden now, showing between 9 and 12 satellites.  With GNSS enabled, this was up around 18-19.

Test flights tomorrow.

giorgos66, 2020-12-05:
> Hi folks. Just an update. It's been chucking it down with rain here for most of the day, so I've managed to get a version of firmware flashed that has the Galileo sat reception disabled (from here: [#17 (comment)](https://github.com/brett8883/DJI_Super-Patcher/issues/17#issuecomment-536314530)).
> 
> I had to do the usual twice flash of stock DJI firmware, before putting this one on. It's sat outside in the garden now, showing between 9 and 12 satellites. With Galileo enabled, this was up around 18-19.
> 
> Test flights tomorrow.

We are waiting for you to tell us your impressions tomorrow. Good flights.

lis-uav, 2020-12-06:
> We are waiting for you to tell us your impressions tomorrow. Good flights.

Just back from some sunrise flights.  Just over an hour of airtime logged, in the middle of nowhere pretty much.  Max satellites was 19.  Zero GPS issues.  I've got some more flights planned for this afternoon, but I'm happy the GPS issue is sorted now.  I didn't test the altitude limit etc.

giorgos66, 2020-12-06:
> Just back from some sunrise flights. Just over an hour of airtime logged, in the middle of nowhere pretty much. Max satellites was 19. Zero GPS issues. I've got some more flights planned for this [...]

Good morning. With the latest Superpatcher 2.0.2  installed and the satellites Galileo? Sorry for the writing but I do it through google translation.

lis-uav, 2020-12-06:
No.  This morning's flights were with the Galileo satellite reception disabled.  The firmware file was from here: #17 (comment).

giorgos66, 2020-12-06:
> No. This morning's flights were with the Galileo satellite reception disabled. The firmware file was from here: #17 (comment). It was installed with Super patcher 2.0.2.

I do not understand what you have done. What specific file did you put together with Superpatcher 2.0.2. There is a lot of discussion there.

lis-uav, 2020-12-06:
I downloaded the firmware bin file from the post linked above.  Then used Superpatcher to connect to my P4ProV1.  Go through and choose whatever for the smart battery mod, and stealth mod.  Do the ADB root access steps.  Then when it comes to flashing the firmware, rather than choosing the firmware bin file Superpatcher says to use, I selected the one I downloaded as above.  That is then flashed to the P4ProV1.

The reason for doing this, is that the standard firmware bin files in Superpatcher, always enable the Galileo satellite reception.  There is currently no option to choose whether this mod is enabled or not.  Hence having to manually download the No-Gal firmware bin file.  From reading the post I linked above, I think this option will possibly be in the next version of superpatcher, when (if?) it is released.

I hope that's clear, but please do ask if anything isn't .

giorgos66, 2020-12-06:
> I downloaded the firmware bin file from the post linked above. Then used Superpatcher to connect to my P4ProV1. Go through and choose whatever for the smart battery mod, and stealth mod. Do the ADB [...]

I understood my friend. All good with the  Superpatcher 2.0.2 except for the satellites Galileo which creates a problem. Did I get it right?

lis-uav, 2020-12-06:
That's correct.  It's only an issue on the Phantom 4 Pro V1 as well.

I am running Superpatcher on a Mavic Pro with Galileo with no issues at all, it's great!

giorgos66, 2020-12-06:
> That's correct. It's only an issue on the Phantom 4 Pro V1 as well.
> 
> I am running Superpatcher on a Mavic Pro with Galileo with no issues at all, it's great!

Too bad because I was interested in the strength of the signal. How do you work with the battery? What adjustment have you made to dji go 4

modarmmmljo, 2024-11-22:
i need P4P_SP_2.0_Fully Loaded__dji_system.bin