brett8883/DJI_Super-Patcher #154: Patcher removal?

smoi, 2021-03-03:
There's no mention on how to remove the patch or if it's safe to update to the latest firmware.
Is it enough to update to the latest firmware to remove the patch?

lovrenco, 2021-08-20:
I would like to know this as well. What is the procedure to flash back factory firmware (without the patch) in case of need?

Weirdoz, 2021-08-20:
Just flash the actual firmware and the patch is gone.

Weirdoz, 2021-08-21:
> There's no mention on how to remove the patch or if it's safe to update to the latest firmware.
> Is it enough to update to the latest firmware to remove the patch?

Yes just flash a "normal" firmware from dji. Or the one you flashed before applying the patch, just flash without patching.