brett8883/DJI_Super-Patcher #174: Windows10 WontLoad

slashster, 2023-10-12:
Super Patcher 2.0.2 on Windows 10, Loads for a few seconds then vanishes. log file below.

START SUPER_PATCHER 
Thu 12/10/2023_14:15:39.22 
I have verified internet access to GitHub 
14:15:42.02

yadangxiaojiajia, 2023-10-29:
where are you? china?

slashster, 2023-11-03:
NZ

Surveying-service-Timash, 2023-11-22:
I have the same problem for older versions.
I'm trying to flash my PSP, I use Super-Patcher version 2.1, but at the moment when SP_Flight_Controllers is installed, the drone does not restart automatically, Dumbledore writes that I can restart the drone. when checking the version through the NLD, the check does not pass (instead of 03.02.44.22, I have 03.02.44.07. I think the thing is that I use Windows 11

RoloF40, 2023-11-22:
Same issue here...

I manage to take a screen shot of the 1st error before the window is closed.. it says: 
C./windows/system32 busybox is not recognized as an internal or external command operable program..

So I copy the .exe to the sys32 folder..
Then the next error is:
cant open 2.0 zip...

WEIRD

RoloF40, 2023-11-22:
same, cant use it... whats wrong?

Adry089, 2024-05-29:
> Super Patcher 2.0.2 su Windows 10, si carica per alcuni secondi e poi svanisce. file di registro qui sotto.
> 
> INIZIO SUPER_PATCHER gio 12/10/2023_14:15:39.22 Ho verificato l'accesso a Internet a GitHub 14:15:42.02

That's right, I have this problem too, but I think we couldn't do anything about it if the creator of it doesn't resurrect

Adry089, 2024-05-29:
> Stesso problema qui...
> 
> Riesco a fare uno screenshot del 1° errore prima che la finestra venga chiusa. dice: C./windows/system32 busybox non è riconosciuto come un programma di comando interno o esterno.
> 
> Quindi copio il .exe nella cartella sys32. Quindi l'errore successivo è: non riesco ad aprire 2.0 zip ...
> 
> STRANO

Oddly enough, it does everyone this way