brett8883/DJI_Super-Patcher #179: super patcher not working

noamel85, 2024-06-17:
when I open super patcher it says the following and closes

Success I have verified internet access to GitHub.

Please wait while I set things up. This wont take long..

***C: \Windows \System32" \busybox.exe" is not recognized as an internal or external command,
executable program or batch file.
****C : \Windows \System32"\busybox.exe" It is not recognized as an internal or external command, program, or executable batch file

Toomanyplanes, 2024-06-18:
I just posted a working fix just before your post. Go get it! Hah!

noamel85, 2024-06-19:
Really.... Can I please send a link
?

maboite3mail, 2024-09-18:
hello, I have the same problem super patcher v2.0.2 opens connects to the network downloads the log file but crashes after. when I restart it it crashes. what should I do? 😅

noamel85, 2024-09-18:
Hi

noamel85, 2024-09-18:
You need to replace a file

noamel85, 2024-09-18:
Send a screenshot of the folder and I will tell you which file to replace

maboite3mail, 2024-09-18:
ok thanks I'll post the screenshot as soon as I get home.

maboite3mail, 2024-09-18:
here are the files
[image]

maboite3mail, 2024-09-18:
[image]

noamel85, 2024-09-19:
Search online for a more recent version of BusyBox and replace the file.
If I remember correctly during installation it replaces the file again.
So even in the middle of installation you will replace the file again.
Let me know if it worked out for you

maboite3mail, 2024-09-19:
a big thank you, I'm happy because it finally works. Thanks to the new downloaded file. thanks again 🤩

noamel85, 2024-10-01:
Glad to hear it!