brett8883/DJI_Super-Patcher #184: Can't calibrate IMU after patching Mavic Pro

clamintus, 2025-05-03:
My Mavic Pro had the latest .500 firmware, so I had to downgrade it first to .400 then .300.
I followed all directives on Super Patcher, enabled Smart Battery Mod and jkson while not selecting Stealth Mode as I don't need it, everything went fine and the aircraft fw version string was blank on DJI Go 4, but it seems I can't fly anymore as it says that it needs IMU recalibration (and also the aircraft seems stuck in Atti mode). It can't complete the calibration no matter what, always fails on step 1/6 with error -50.
I've tried calibrating from DJI Assistant and repatching the drone through the super patcher. Everything was done successfully but the problem persists.
I'll gladly send any log if needed

clamintus, 2025-05-03:
Solved it by flashing stock again and recalibrating in a different position than the one shown by the calibration process

closed 2025-05-03 by clamintus

audiguy88, 2025-05-20:
😐🤮